jyjblrd/Low-Cost-Mocap #4: Remove cv.sfm dependency

jyjblrd, 2023-11-21:
This project depends on the OpenCV SFM (structure from motion) module, which requires you to build OpenCV from source. This is a huge pain, especially since we only use three simple functions from the module: `fundamentalFromProjections`, `essentialFromFundamental`, and `motionFromEssential`.

It should be very easy to reimplement these functions in python by looking at the module source code, and this would remove `cv.sfm` as a dependancy, making the project a lot more accessible.

SANTHOSH-MAMIDISETTI, 2023-11-22:
@jyjblrd  , Could you link where those functions are present in the [OpenCV](https://github.com/opencv/opencv) repo , I would like to have a look and start

jyjblrd, 2023-11-22:
Hey @SANTHOSH-MAMIDISETTI, you can find the occurrences [here](https://github.com/search?q=repo%3Ajyjblrd%2FMocap-Drones%20cv.sfm&type=code). It would be amazing if you can do this!

The source code for the sfm module is [here](https://github.com/opencv/opencv_contrib/tree/4.x/modules/sfm), and [heres](https://github.com/search?q=repo%3Aopencv%2Fopencv_contrib+path%3A%2F%5Emodules%5C%2Fsfm%5C%2F%2F+fundamentalFromProjections&type=code) an example of searching up the implementation of the `fundimentalFromProjections` function.

NightHawk32, 2023-11-29:
@jyjblrd could you provide test data of **camera1_image_points** and **camera2_image_points** like used [here](https://github.com/jyjblrd/Mocap-Drones/blob/7857ac0c0918948846a084a2c50841a76bc385dd/computer_code/api/index.py#L246)?
If possible the output of **F** and **E** would also be great. I am not entirely sure what the camera points look like. Additionally the test data would help to compare the openCV implementation of essentialFromFundamental to the python version.

jyjblrd, 2024-05-22:
Resolved by [https://github.com/jyjblrd/Low-Cost-Mocap/pull/42](https://github.com/jyjblrd/Low-Cost-Mocap/pull/42), which has been pulle into the [`no-cv-sfm`](https://github.com/jyjblrd/Low-Cost-Mocap/tree/no-cv-sfm) branch. However, this code is currently untested – test cases will need to be written before this issue can be closed.

high993, 2024-09-10:
Hey, @jyjblrd I have an amazing project in the works for 3D Markerless Human-Mocap using AI Models, that's very similar to your project.  I plan on pushing the project to github some time in the next days or weeks. In the course of making it, I learned that to avoid using opencv sfm some concessions must be made, for starters, fundamental matrix generators like RANSAC and LMEDS need a minimum number of points (8) to solve its position (or 7 if you make a few assumption's). I circumvented this by running my extrinsic calibration off of a large checker pattern placed in the center of the scene. However since I haven't made the IR filter modifications to the camera and I'm running the cameras in the res.LARGE (640x480) configuration, I'm not sure if this solution will work for you due to the compute overhead.  Also, I plan on making a YT vid explaining my code and I felt like that a proper reference to yourself was in order. Id be happy to hash out the details of the absence of opencv sfm after its published. 

Cheers! 
Adam,

PR #42 (2024-05-22): Remove cv.sfm dependency
Closes #4.

I decided to take this on to better understand the math going on in the backend here. I haven't built the sfm version of opencv myself, and don't have the pseye cameras or drones to test my replacement of it myself, so thorough real use testing should be done before merging this.

I did look at the opencv_sfm test suite, but they only test by making raw data, doing one set of conversions, converting back, and testing that the results match. Since I didn't implement the reverse of these functions I didn't implement that, and it did not give me examples to te [... cut]