jyjblrd/Low-Cost-Mocap #9: Serial Port needs to be configured

Gadgeteering, 2023-11-22:
When running api/index.py the serial port need to be configure to the serial controlling the quadcopter line 21 otherwise you get the error below

`serial.serialutil.SerialException: [Errno 2] could not open port /dev/cu.usbserial-02X2K2GE: [Errno 2] No such file or directory: '/dev/cu.usbserial-02X2K2GE'`

jyjblrd, 2023-11-22:
Yeah, that should definitely be configured some other way. Perhaps a command line argument or even a dropdown on the front end.

AliHusseinAlmoussawi, 2024-08-14:
Hello, is it necessary to configure the serial port to serial device controlling the quadcopter in order to run the command api/index.py?? Also I want to ask if this step is necessary for the cameras to stream and display at the frontend because I have run the frontend, but there is no displaying for the cameras.