jyjblrd/Low-Cost-Mocap #29: python3 api\index.py failed

Blaise-wig, 2024-03-12:
Hello, when I run python3 api\index.py, I get this error

(DroneControl) C:\Users\Victus\anaconda3\envs\DroneControl\Mocap-Drones-main\computer_code>python api\index.py
 * Serving Flask app 'index'
 * Debug mode: on
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:3001
Press CTRL+C to quit
 * Restarting with stat
Traceback (most recent call last):
  File "C:\Users\Victus\anaconda3\envs\DroneControl\Mocap-Drones-main\computer_code\api\index.py", line 22, in <module>
    ser = serial.Serial("COM12", 115200, write_timeout=1, )
  File "C:\Users\Victus\anaconda3\envs\DroneControl\lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\Victus\anaconda3\envs\DroneControl\lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Users\Victus\anaconda3\envs\DroneControl\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM12': PermissionError(13, 'Access is denied.', None, 5)

I connected the esp32 to COM12 and made sure no other program is using the port. I have been trying to resolve this for two days now. can anyone help?

jyjblrd, 2024-03-12:
`PermissionError(13, 'Access is denied.', None, 5)`
 have you tried running with sudo? ie. `sudo python3 api\index.py`

jyjblrd, 2024-03-12:
Also make sure that your arduino ide / whatever you are using to program the ESP32 is closed, because it could also be connecting to the port.

Blaise-wig, 2024-03-13:
I'm on windows. But I ran it as administrator. I also closed the arduino IDE but the error still persisted.

clennpillo, 2024-05-26:
weird windows permission problem, I gave up windows and tried WSL, it worked.

clennpillo, 2024-06-08:
I was lately trying to run the code on windows10, and solved this permission problem, here is the detail.

if you are quite sure you serial port is not occupied by another program which is the situation in my case, but you still ran into the weird permissionerror 13 problem, you may try these steps.

first, turn off the reloader function of Flask-SocketIO, as flask-socketio might use your serial port which caused the permission problem.

second, create a unique thread for pyserial to operate serial port.

here is the code i use

`
app = Flask(__name__)
socketio = SocketIO(app, cors_allowed_origins='*', async_mode='threading')

serial_lock = threading.Lock()
ser = None

def serial_worker():
    global ser
    try:
        with serial_lock:
            ser = serial.Serial("COM5", 1000000, write_timeout=1)
            while True:
                data = ser.read(100)  # 示例串口读取操作
                if data:
                    print(f"Received data: {data}")
    except Exception as e:
        print(f"Serial exception: {e}")
`

`
if __name__ == '__main__':

    serial_thread = threading.Thread(target=serial_worker)
    serial_thread.daemon = True  # 让线程在主程序退出时自动结束
    serial_thread.start()

    socketio.run(app, port=3001, debug=True, use_reloader=False)
`

closed 2024-07-12 by jyjblrd