jyjblrd/Low-Cost-Mocap #32: error compiling cython file

986164283, 2024-04-12:
hello
I got an error when I try to install python setup.py install. I'm on Ubuntu 22. the main error:

Error compiling Cython file:
------------------------------------------------------------
...
        # init context
        print("about to init")
        try:
            ps3eye_init()
        except Exception as exc:
            print exc
                  ^
------------------------------------------------------------

pseyepy/cameras.pyx:245:18: Syntax error in simple statement list
Traceback (most recent call last):
  File "/root/pseyepy-master/setup.py", line 76, in <module>
    ext_modules=cythonize(extensions),)
                ^^^^^^^^^^^^^^^^^^^^^
  File "///anaconda3/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
    cythonize_one(*args)
  File "//anaconda3/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: pseyepy/cameras.pyx

The above is where the error is reported

I hope someone can help  and give me more information

Thanks in advance.

jyjblrd, 2024-04-12:
Hi, that seems to be a bug in the pseyepy library: github.com/bensondaled/pseyepy/

Something is going wrong in the `ps3eye_init()` function and the exception isnt being printed because it should be `print(exc)` not `print exc`. I would change this line so you actually can see what the error is to properly debug the problem, and then reinstall the library.

jyjblrd, 2024-04-12:
This issue looks to be the same issue that you had: [https://github.com/bensondaled/pseyepy/issues/15](https://github.com/bensondaled/pseyepy/issues/15)

986164283, 2024-04-15:
Hi, have you modified it? I downloaded and reinstalled and still got the same prompt。

jyjblrd, 2024-04-15:
Hi, you should reinstall the pseyepy library, modify the code as I mentioned, and try run it again. Or you could pull [this branch](https://github.com/bensondaled/pseyepy/pull/18) which looks to solve the problem.

986164283, 2024-04-16:
Hello, after following the modifications you mentioned, the problem has been solved, thank you.

林玮
***@***.***

closed 2024-04-23 by jyjblrd

luislglp, 2024-05-31:
Thanks for mentioning the checklist, jyjblrd.
I still had some issues with the pseyepy library even after following zalo's checklist to install on Windows 11 (VS 2022).
Maybe this is too basic, but not sure if we need somehow to install the camera drivers first.
I downloaded the CL-Eye driver, tested the cameras and after that the pseyepy library worked fine.