jyjblrd/Low-Cost-Mocap #39: "python3 api\index.py" successful, but the video stream is not visible on the UI.

cultist123, 2024-05-17:
File "/Users/apple/Documents/2 工作空间/git/Low-Cost-Mocap-main/computer_code/api/index.py", line 55, in gen
    frames = cameras.get_frames()
  File "/Users/apple/Documents/2 工作空间/git/Low-Cost-Mocap-main/computer_code/api/helpers.py", line 138, in get_frames
    frames = self._camera_read()
  File "/Users/apple/Documents/2 工作空间/git/Low-Cost-Mocap-main/computer_code/api/helpers.py", line 72, in _camera_read
    frames[i] = np.rot90(frames[i], k=self.camera_params[i]["rotation"])
ValueError: assignment destination is read-only

The source code may have issues.

jyjblrd, 2024-05-17:
That is strange. Try adding the line `frame[i] = frame[i].copy()` before the line that is causing the error.

cultist123, 2024-05-21:
The issue has been resolved, and the software can now be used by simply plugging in multiple cameras. It is recommended to have all hardware connections completed before running the software.