jyjblrd/Low-Cost-Mocap #43: Battery 9A seems way too big for the boards

clennpillo, 2024-05-22:
hi,

I found 9A battery is too big for the boards, already ruined ESP32 receiver, and F3 EVO is very hot while working, I'm not sure if I had done it right though.

jyjblrd, 2024-05-22:
The battery shouldn't be blowing up your boards.
I connected the battery to the F3 EVO flight controller, and used the 3.3v output from that board to power the ESP32.

Perhaps you installed the battery with the wrong polarity (ie. mixed up +/-).. There is no protection against this and even doing it once can destroy the boards.

clennpillo, 2024-05-23:
Maybe you're right, anyway I just used a smaller battery that is 1A, hope it works well.

closed 2024-05-23 by clennpillo

jyjblrd, 2024-05-23:
Amperage does not have any impact on if a battery will blow up a board, it's the voltage that kills a board. Single cell Li-ion batteries are always around 3.7v (because of chemistry), which the flight controller is built to handle.

The amp-hour (Ah) rating of a battery is just how long it will last. It's like how putting a larger fuel tank in your car won't damage the engine... if that makes sense. 

Just don't reverse the polarity and you'll be fine.

clennpillo, 2024-05-24:
I guess my board blowed up was because I once reversed the polarity. I always think Amperage does not have any impact on if a battery will blow up a board, if I use 3.7V it's always ok.

one question is that why the battery has a paramter of 9A, what does this indicates ?

jyjblrd, 2024-05-24:
The battery I recommend is the 18350 type (18mm * 35mm). The one I used has a voltage of 3.7v, a capacity of 1100mAh, and a max current if 9A. The voltage is self explanatory. 

1100mAh means that it can output 1100milliamps (at 3.7v) for 1 hour (this works out to 1.1Ah * 3.7v = 4.07watthours. 

9A means that the max current it can provide is 9 amps, or 9A*3.7V = 33.3 watts.

clennpillo, 2024-05-30:
I used the 9A battery again, the F3 board was again much hotter than the small battery, I then found this picture below in your video, it showed that you used two resistances on the board, is this the reason that the current on your board is not that high.

[image]

jyjblrd, 2024-05-30:
No, those resistors are a voltage divider to measure the voltage level of the battery (which I never wrote the code for). 

After reversing the polarity of the F3 board, it will be permanently damaged, you may need to buy a new one.