jyjblrd/Low-Cost-Mocap #44: ESP32 receiver soldering with F3 EVO

clennpillo, 2024-05-26:
hi,

I got a problem that I dont' know how to soldering ESP32 receiver to F3 EVO board, I watched the video but there was no details about this.

My betaflight is as follow
[image]

### ESP32 receiver soldering
I suppose four terminals need to be soldered.
[image]

### F3 EVO soldering
I know how to solder power terminals, but for signals, because my betaflight showing that UART2's serial function is enabled as in the screen shot above, I guess I need to use UART2's two terminals, one is TX, another is RX, however, I couldn't find any article or video that explain which terminals are for UART1, UART2, UART3.
[image]

I tried to connect in the way above before soldering, but it turned out the motors won't start, so I think there might be some problem here.

Can you tell me where can I find this information about UART1,2,3 's corresponding terminals, and can you give me your sodering picture ?

cultist123, 2024-05-26:
Although I haven't solved this problem for you, I have done some research on it. The F3 board does not have the corresponding communication pins, so you can only solder it directly to the STM32 chip. I suggest you refer to the STM32 chip diagram as shown below, and solder it as demonstrated in the video.
[image]

jyjblrd, 2024-05-26:
You actually need to connect the ESP32 to the UART 3. UART 1 is used by the usb interface (i think?) and UART 2 is used by the onboard RC receiver (useful to keep around for manual flying / debug purposes).

[This location in the video](https://www.youtube.com/watch?v=0ql20JKrscQ&t=157s) shows the UART 3 RX pin to solder to. You can then tell betaflight to use UART 3 as the RC signal. It isn't the best solution, a better flight controller that actually breaks out the extra UART as a solderable pin would be better.

clennpillo, 2024-05-27:
Thanks for jyjblrd and cultist123's replies which helps me out.

now I get it, the right terminal is on STM32 instead of F3 board. I just tried to connect as follows.

[image]

on the ESP32 receiver side, I deployed React frontend, python backend ESP32 sender and ESP32 receiver, the receiver was able to send data via sbus and I could see the logs while I click buttons at React frontend, I changed some code slightly to let the code run even if the cameras are not ready yet. The screenshot is what I see in Arduino running ESP32 receiver.
[image]

I clicked arm at frontend, but motors still won't start. Any suggestions ?

clennpillo, 2024-05-27:
I did some debug, when I type status command in CLI of betaflight, I got RXLOSS which means the receiver is not ready, didn't operation connecting ESP32 to STM32 make a good receiver for the F3 EVO board ? 

[image]

jyjblrd, 2024-05-27:
Yes, RX loss means that the flight controller is not receiving any RC signal from the ESP32.
1. Are you sure that the cable is connected correctly to the flight controller's pin? It is a very very small pin as I'm sure you know.
2. Is the ESP 32 receiver's code configured to send the data out of the correct digital out pin?
3. Is Betaflight correctly configured to receive the SBus data from the ESP 32?

clennpillo, 2024-05-27:
Turned out I didn't make the tiny pin from STM32 connection right.

I did it again and now the RXLOSS problem is solved, so I guess I know the correct pair that should be soldered between F3 EVO and ESP32 receiver, however, I tried many many times to solder the tiny pin and failed many many times, that's really the most difficult part in this project, till now I 'm still trying.

jyjblrd, 2024-05-27:
Yeah, a better flight controller which actually breaks out the extra UART would make things a lot easier. Better yet, the esp32 could be the flight controller itself (one core receiving and decoding messages, another core controlling the motors).

clennpillo, 2024-05-27:
This is a great project that exactly meet our needs, so I tried my best to use what hardware and software you use in order to understand this project, after making this right, we gonna make changes and upgrade,etc.  Really appreciate what you have done.