jyjblrd/Low-Cost-Mocap #45: About compiling SFM module

cultist123, 2024-05-26:
Based on macOS 14 system, after compiling opencv and opencv_contrib, I found that there is no 'sfm' module in 'local/include/opencv4/opencv2', even though I checked the 'sfm' module in CMake. Were you compiling it on Ubuntu? If not, could you please tell me the process of compiling it on macOS? Do I need to install sfm dependencies first and then compile and download using CMake?

jyjblrd, 2024-05-26:
I compiled it on my M2 Mac.

The following links may be helpful:
[https://hackyourcode.medium.com/how-to-build-and-install-opencv-4-x-on-macos-3a2ce4d82eb4](https://hackyourcode.medium.com/how-to-build-and-install-opencv-4-x-on-macos-3a2ce4d82eb4)
[https://github.com/opencv/opencv_contrib](https://github.com/opencv/opencv_contrib)

May want to add only sfm by adding the flag to the cmake command: `-DBUILD_LIST="sfm"`.

If you're feeling adventurous, you can try use the `no-cv-sfm` branch of this repo, which has removed the sfm module dependency. However, it is completely untested, so it might not work at all (bug fixes would be welcome!)

cultist123, 2024-05-26:
Are you configured through a GUI interface or command line? Even after compiling and downloading the library, I still encounter the error 'module 'cv2' has no attribute 'sfm'.' I've been stuck on this for a long time and can't seem to resolve it. Could you please advise on how to fix this?

jyjblrd, 2024-05-26:
I used the command line.

If your python interpreter is able to find the cv2 module but not the sfm module within it, it means that your old cv2 install is still present. Try use pip to remove cv2 to make sure that the cv2 module is the one being built from source.

cultist123, 2024-05-31:
when i remove the old cv2 and make sure the cv2 module is  built from source, i am unable to find the cv2 module.

jyjblrd, 2024-05-31:
That means that you are not properly installing the opencv library which you compiled from source. Unfortunately I can't really help you with this, there are plenty of resources online which explain how to compile and install opencv from source who understand the process better than I do.

closed 2024-07-12 by jyjblrd

clennpillo, 2024-07-16:
I used cmake-gui.

You probably need to confirm that sfm appears in the correct place in cmake-gui tool after you click configure, you see it, you got sfm in the compiled opencv.