jyjblrd/Low-Cost-Mocap #56: Openning 2 or more webcams is not possible ?

clennpillo, 2024-07-08:
hi jyjblrd,

I ran into a problem that I cannot open 2 or more webcams at the same time on my computer. Opening first webcam is ok, but openning the second one fails. This happens when I plug in 2 or more webcams on my USB hub which connects to one USB3.0 port of my computer, however, if I plug in 2 webcams on 2 seperate USB ports of my computer, everything works fine, the error information is as follows.

when using openCV, 
ret, frame = cap.read() always gets a False for ret which means openning failure

I also tried VLC, the error was
capture device might be in use

This is some information for reference.
laptop: Thinkpad L14
os: windows10 home edition 22H2
USB hub: external power supply and USB3.0 support

Can you tell me how you connect 4 cameras to your computer in your instance ?

jyjblrd, 2024-07-12:
Hi, 

Has this issue been solved? (and if so, please let others know how you fixed it in case others have the same issue!)

I vaguely remember having issues if I plugged too many cameras into the same cheap USB hub, but had no issues with my powered usb hub.

clennpillo, 2024-07-16:
I used night vision usb cameras, I never tried pseye. Pseye python library must have dealt with multi-camera problem as long as you use a powered hub.

I use Windows 10,  I got 2 USB controller ports on my computer, one connected a cheap usb hub, plugged in two cameras, the other connected a powered usb hub, plugged in other two cameras.

I adjusted camera parameter in code that make resolution to 320x240, this make sure you don't have bandwidth issue.

I started 4 individual processes to read camera frames, started a websocket server, the camera readers sent frame data to websocket server, I added some code to read from websocket server to get frames in helpers.py.

jyjblrd, 2024-07-16:
Hmm, not sure about how windows handles multiple webcams but I assume it should be fine? Unfortunately I can't help much as I don't have a windows computer or multiple webcams.