jyjblrd/Low-Cost-Mocap #62: Socket.IO communication issue

lurz01, 2024-08-07:
Hello,

Thank you for contributing such a powerful open-source project. I followed your instructions to run the project and found that the video stream is transmitted normally, but the Socket.IO communication is not working properly. Have you encountered a similar issue, or could you provide some troubleshooting ideas? I would greatly appreciate it.

deyoyow, 2024-09-30:
Hello lurz, may i ask is your flask running on the index.py

`if __name__ == '__main__':
    socketio.run(app, port=3001, debug=True)`
    
 Mine seems to kept re-opening the COM port if the parameter set to `debug=True`
 and while `debug=False` resolves the error, the camera is seems to be not running (FPS = 0)
[image]

cultist123, 2025-01-12:
the solution is to connect four cameras