jyjblrd/Low-Cost-Mocap #63: Error while building OpenCV

zabsy, 2024-08-08:
Hey Joshua Bird,

I'm building opencv from source on Ubuntu 22.04.4, following the instructions [here](https://docs.opencv.org/4.x/d7/d9f/tutorial_linux_install.html?ref=wasyresearch.com) exactly, except replacing "4.x" with "4.10.0" 

After running the configuration step: 
`cmake -DOPENCV_EXTRA_MODULES_PATH=../opencv_contrib-4.10.0/modules ../opencv-4.10.0`

I'm running into these errors:
```
CMake Error at cmake/OpenCVUtils.cmake:1582 (add_library):
  Target "opencv_sfm" links to target "absl::log" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/OpenCVModule.cmake:989 (ocv_add_library)
  cmake/OpenCVModule.cmake:905 (_ocv_create_module)
  /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/CMakeLists.txt:150 (ocv_create_module)

CMake Error at cmake/OpenCVUtils.cmake:1582 (add_library):
  Target "opencv_sfm" links to target "absl::check" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/OpenCVModule.cmake:989 (ocv_add_library)
  cmake/OpenCVModule.cmake:905 (_ocv_create_module)
  /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/CMakeLists.txt:150 (ocv_create_module)

CMake Error at cmake/OpenCVUtils.cmake:1582 (add_library):
  Target "opencv_sfm" links to target "absl::fixed_array" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/OpenCVModule.cmake:989 (ocv_add_library)
  cmake/OpenCVModule.cmake:905 (_ocv_create_module)
  /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/CMakeLists.txt:150 (ocv_create_module)

CMake Error at /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/src/libmv_light/libmv/correspondence/CMakeLists.txt:9 (ADD_LIBRARY):
  Target "opencv.sfm.correspondence" links to target "absl::log" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/src/libmv_light/libmv/correspondence/CMakeLists.txt:9 (ADD_LIBRARY):
  Target "opencv.sfm.correspondence" links to target "absl::check" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/src/libmv_light/libmv/correspondence/CMakeLists.txt:9 (ADD_LIBRARY):
  Target "opencv.sfm.correspondence" links to target "absl::fixed_array" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/src/libmv_light/libmv/multiview/CMakeLists.txt:19 (ADD_LIBRARY):
  Target "opencv.sfm.multiview" links to target "absl::log" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/src/libmv_light/libmv/multiview/CMakeLists.txt:19 (ADD_LIBRARY):
  Target "opencv.sfm.multiview" links to target "absl::check" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at /home/zabsy/opencv_build/opencv_contrib-4.10.0/modules/sfm/src/libmv_light/libmv/multiview/CMakeLists.txt:19 (ADD_LIBRARY):
  Target "opencv.sfm.multiview" links to target "absl::fixed_array" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
```

I tried to fix it by installing `sudo apt-get install libabsl-dev` but I still end up with the same errors. 

Did you encounter any similar issues, or do you have any ideas for how to fix this? Any suggestions would be greatly appreciated. 

Also, 
This project is amazing, thank you for making this open source!

cultist123, 2025-01-12:
You need to go to opencv's GitHub to ask this question. This project has many configuration points, which is relatively cumbersome. My solution is to find professionals to configure the sfm module for me online.

deyoyow, 2025-03-06:
Building OpenCV from scratch is a cumbersome task on Windows, sadly I forgot to document my exact step but here is my CMake Configuration

set CMAKE_BUILD_PARALLEL_LEVEL=2
"C:/Program Files/CMake/bin/cmake.exe" -H"C:/Users/LENOVO/Desktop/Python/Swarm Track/opencv" ^
    -B"C:/Users/LENOVO/Desktop/Python/Swarm Track/opencv/build" -G"Visual Studio 17 2022" ^
    -DINSTALL_TESTS=ON ^
    -DINSTALL_C_EXAMPLES=ON ^
    -DBUILD_opencv_python3=ON ^
     ^
    -DBUILD_EXAMPLES=ON ^
    -DBUILD_opencv_world=ON ^
    -DGlog_LIBRARY="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/glog/install/lib/glog.lib" ^
    -DGflags_LIBRARY="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/gflags/install/lib/gflags.lib" ^
    -DPYTHON3_INCLUDE_DIR=C:/Users/LENOVO/anaconda3/envs/swarm-track/include ^
    -DPYTHON3_LIBRARY=C:/Users/LENOVO/anaconda3/envs/swarm-track/libs/python39.lib ^
    -DPYTHON3_EXECUTABLE=C:/Users/LENOVO/anaconda3/envs/swarm-track/python.exe ^
    -DPYTHON3_NUMPY_INCLUDE_DIRS=C:/Users/LENOVO/anaconda3/envs/swarm-track/lib/site-packages/numpy/core/include ^
    -DPYTHON3_PACKAGES_PATH=C:/Users/LENOVO/anaconda3/envs/swarm-track/Lib/site-packages/ ^
    -DGLOG_INCLUDE_DIR="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/glog/install/include" ^
    -DGlog_DIR="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/glog/install/lib/cmake/glog" ^
    -DGFLAGS_INCLUDE_DIR="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/gflags/install/include" ^
    -DGflags_DIR="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/gflags/install/lib/cmake/gflags" ^
    -DCERES_DIR="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/ceres-bin/install/lib/cmake/Ceres" ^
    -Dgflags_DIR="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/gflags/install/lib/cmake/gflags" ^
    -Dglog_DIR="C:/Users/LENOVO/Desktop/Python/Swarm Track/ceres/glog/install/lib/cmake/glog" ^
    -DOPENCV_EXTRA_MODULES_PATH="C:/Users/LENOVO/Desktop/Python/Swarm Track/opencv_contrib/modules" ^
    -DBUILD_opencv_sfm=ON ^
    -DCERES_FOUND=ON