jyjblrd/Low-Cost-Mocap #66: [BUG] Issue with running the index.py file

thesis09, 2024-08-17:
**Describe the bug**
the pseyepy is having issue says :

File "M:\Low-Cost-Mocap-main\computer_code\api\index.py", line 1, in <module>
    from helpers import camera_pose_to_serializable, calculate_reprojection_errors, bundle_adjustment, Cameras, triangulate_points
  File "M:\Low-Cost-Mocap-main\computer_code\api\helpers.py", line 12, in <module>
    from pseyepy import Camera
ModuleNotFoundError: No module named 'pseyepy'

but after some research i found the repo and cloned it but still i am encountering the same issue 
when i type (pip install .)
it gives me this new issue:

Processing m:\low-cost-mocap-main\pseyepy
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      M:\Low-Cost-Mocap-main\pseyepy\setup.py:48: UserWarning: Setup params not yet fully tested for Windows.
        warnings.warn('Setup params not yet fully tested for Windows.')
      C:\Users\MAG\AppData\Local\Programs\Python\Python311\Lib\site-packages\Cython\Compiler\Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: M:\Low-Cost-Mocap-main\pseyepy\pseyepy\cameras.pyx
        tree = Parsing.p_module(s, pxd, full_module_name)

      Error compiling Cython file:
      ------------------------------------------------------------
      ...
              # init context
              print("about to init")
              try:
                  ps3eye_init()
              except Exception as exc:
                  print exc
                        ^
      ------------------------------------------------------------

      pseyepy\cameras.pyx:245:18: Syntax error in simple statement list
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "M:\Low-Cost-Mocap-main\pseyepy\setup.py", line 76, in <module>
          ext_modules=cythonize(extensions),)
                      ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\MAG\AppData\Local\Programs\Python\Python311\Lib\site-packages\Cython\Build\Dependencies.py", line 1154, in cythonize
          cythonize_one(*args)
        File "C:\Users\MAG\AppData\Local\Programs\Python\Python311\Lib\site-packages\Cython\Build\Dependencies.py", line 1321, in cythonize_one
          raise CompileError(None, pyx_file)
      Cython.Compiler.Errors.CompileError: pseyepy/cameras.pyx
      Compiling pseyepy/cameras.pyx because it changed.
      [1/1] Cythonizing pseyepy/cameras.pyx
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details

after fixing it , still i am unable to run the backend 
Can you please help me in this ...

deyoyow, 2024-09-25:
Hey, are you able to run the pseyepy stand alone?

If not please try this to run it independently first with the code provided at the repo:

`from pseyepy import Camera, Display

c = Camera() # initialize a camera
d = Display(c) # begin the display`

It works fine for me after following the instructions at 
[https://github.com/bensondaled/pseyepy/issues/22](url) 

Cheers,
Deyo

deyoyow, 2024-09-30:
Hey, sorry for the broken link you could copy the raw text instead. 

[Here](https://github.com/bensondaled/pseyepy/issues/22)

Cheers, Deyo