jyjblrd/Low-Cost-Mocap #69: Camera capture broken when trying to start camera stream

deyoyow, 2024-09-30:
So I tried to run 'python api/index.py' and when I started the stream it looks like this:
[image]

I've succesful in tackling the COM port error in Windows following this  #29 
camera works fine if I try with the original pseyepy command

Is it really necessary to have 4 cameras plugged in for the stream to work as mentioned in #39?

Regards,
Deyo

jyjblrd, 2024-10-01:
Are there any errors printed by the python program?

Does the python program identify the correct number of cameras? (see this line: https://github.com/jyjblrd/Low-Cost-Mocap/blob/8ac4d4107fda026d787d5e197f0dcbe6257b2720/computer_code/api/helpers.py#L26)

cultist123, 2024-11-21:
I encountered the same issue as you, and it requires connecting four cameras to resolve it.

RayInst, 2025-03-03:
The ps3eye driver doesn't support the simultaneous use of more than two ps3eye cameras very well.

deyoyow, 2025-03-06:
> I encountered the same issue as you, and it requires connecting four cameras to resolve it.

Do you use 4 USB Cable with a hub into a single port? or do you mind explaining your setup for it to work?

Regards
Deyo

lamp-deng, 2026-03-01:
I encountered the same issue as you,in the terminal,你会看到摄像头进程被杀了，运行完index.py回来就发现摄像头掉了，虽然print(self.num_cameras) 这个的数量一直是2没变，然后我打印了一下print(cameras_init)的值，显示没有初始化成功，程序都没有进入到camera_stream（）这个函数里面，不太妙啊，