jyjblrd/Low-Cost-Mocap #78: IR floodlight ring

Mohamed-magdy22, 2025-03-30:
Do you know if the IR floodlight is essential for use with the camera? I plan to use only one drone with IR LED markers, as I cannot obtain the IR floodlight ring in my country.

thanks.

janlishak, 2025-04-20:
Hi, the IR floodlights are optional. However, they might make testing and debugging a little easier. Here's a list of things that will be affected.

Camera intrinsics calculation
After you put the IR passthrough filter (floppy disk) on the cameras, they won't be able to capture the checkerboard pattern. You would need to illuminate them with the IR floodlight. To calibrate without the IR floodlight, you'll have to do the calibration without the filters, which might introduce a small inaccuracy since the filters can distort the image a little, but it probably won't be a problem. To sum up, make sure you get the intrinsics of the camera before you put the filters on.

Everything will be dark
Since there will not be any IR light in your room, it might be harder to orient the cameras. Just make sure none of the cameras are obscured by furniture or other objects in the room because you won't be able to actually see it in the image.

Retroreflective markers
As far as I can tell, Joshua intended to use the IR floodlights with retroreflective markers. The advantage would be that the markers wouldn't need to be powered because they would reflect the light from the IR floodlights. But if you want to use IR LEDs, as Joshua did, it might be actually better to turn the floodlights off to minimize noise.

Mohamed-magdy22, 2025-04-21:
Thank you very much—I truly appreciate your support.

 I have one more question: Is it necessary to calibrate the cameras using a traditional chessboard pattern, or is it also acceptable to use infrared (IR) markers as an alternative method for calibration?

janlishak, 2025-04-22:
Hi, using the checkerboard pattern is the standard way to estimate the intrinsics of a camera. It is already implemented in OpenCV, and the steps are discussed in this thread: https://github.com/jyjblrd/Low-Cost-Mocap/discussions/11

In my case, I was able to use some values I found in the thread. All PS3 cameras should be quite similar, so if you're lucky, you might be able to use someone else's values if they had a similar setup. However, getting the actual values for your own cameras is definitely recommended. With incorrect intrinsics, you'll have trouble estimating the poses when running the project.

Mohamed-magdy22, 2025-04-23:
Thank you very much—I truly appreciate your support.

closed 2025-05-03 by Mohamed-magdy22